adafruit/Adafruit_ADS1X15 #12: Not configured for use with ADS1115

adigregorio, 2015-10-08:
According to the ADS1115 datasheet, the maximum sample rate for the ADS1115 is 860 SPS. However, there is no way to change the sample rate in your code, and the default is set to 1600 SPS, as noted in Line 87 of ADS1015.h.

soligen2010, 2016-01-04:
For the ADS 1015 it is 1600 SPS (the default), but the same register bit pattern on the ADS 1115 (according to the data sheet) equates to 128 SPS (also the default).  So with no modifications this library will sample at 128 SPS for the 1115 and 1600 SPS for the 1015.  I agree it would be nice to be able to change this, as it could be faster for both chips, but the library is within specs for both chips.

chaeplin, 2016-01-08:
I have check datasheet, both chips use Bits[7:5], DR[2:0].

    #define ADS1015_REG_CONFIG_DR_MASK      (0x00E0)  
    #define ADS1015_REG_CONFIG_DR_128SPS    (0x0000)  // 128 samples per second  // 8SPS ADS1115
    #define ADS1015_REG_CONFIG_DR_250SPS    (0x0020)  // 250 samples per second // 16SPS
    #define ADS1015_REG_CONFIG_DR_490SPS    (0x0040)  // 490 samples per second // 32SPS
    #define ADS1015_REG_CONFIG_DR_920SPS    (0x0060)  // 920 samples per second // 64SPS
    #define ADS1015_REG_CONFIG_DR_1600SPS   (0x0080)  // 1600 samples per second (default) // 128SPS(default)
    #define ADS1015_REG_CONFIG_DR_2400SPS   (0x00A0)  // 2400 samples per second // 475SPS
    #define ADS1015_REG_CONFIG_DR_3300SPS   (0x00C0)  // 3300 samples per second // 860SPS

soligen2010, 2016-01-08:
I am working on a new version of the library that includes the ability to set the SPS, and other good changes, some of which were already done by other people, but the pull requests were never merged back into the master.  I probably will have to create a new fork to hold a version with all the changes.  If you are interested in helping to test the new fork, please let me know.

ybizeul, 2017-01-23:
That library could definitely use some improvements. The header file especially is very confusing and it shouldn't be hard to implement the read delay according to the SPS.

@soligen2010 thanks for the work on the fork!

soligen2010, 2017-01-23:
You are welcome for the work - I am happy others find it useful.

Since it has been over a year on this thread, I'll give an update:  my fork supports all SPS setting and uses polling instead of fixed timing so it works more reliably with clone chips.  Also, number of other functionality improvements that more fully support the capabilities of the chip.

I have also since then bought an ADS1015, so it has also been tested with both chips.

If you have other improvements to make, please code them against my fork so they can be merged.

for others looking for my fork, it is here: [https://github.com/soligen2010/Adafruit_ADS1X15](https://github.com/soligen2010/Adafruit_ADS1X15)

ybizeul, 2017-01-23:
Because of that I'm actually using https://github.com/jrowberg/i2cdevlib as they had a standard library package for platformio. It seems they also use a polling logic, but for some reason, I couldn't get better than 3.7ms for each poll, now I'm not sure exactly how much of the 3.7 is my code or the library code

soligen2010, 2017-01-24:
I think I'm getting 1 to 2 milliseconds best case taking single shot reading, but I am trying to recall from a year ago when I tested polling.  Using single shot readings you will not be ale to achieve the full SPS in your code because there is time overhead the chip needs to get started and communication time for the i2c commands, etc.  To get the full SPS you need to use continuous mode and use the interrupt pin (example program is in my fork).  The downside to continuous is that you can only use one input (either one single ended or one differential).  If you want to use multiple inputs at the same time, then the single shot is what you need to do unless you want to use a different chip for each input.

Also, some chips (presumably clones) are slower than than others.  I have a small sample set, but for me the small purple boards from e-bay sellers are slower than the blue boards (this may not be universally true)

magicChristian, 2019-04-11:
Hallo Solingen,  I tried to set the SampleRate but with   ads.setSPS(ADS1115_DR_16SPS);   I got an error message "class Adafruit_ADS1115" has no member named "setSPS"
Is there already a library which allows to set the samplerate? Thanks for your an answere. Christian

RenatoTorres, 2019-05-15:
Dear all,

Where is support for all ADS1115 data rates and ADS1115 other features?

Does someone have any other lib to suggest?

Thanks a lot,

Renato

RenatoTorres, 2019-05-15:
I found, sorry!
https://github.com/soligen2010/Adafruit_ADS1X15

HamishB, 2019-07-25:
Hi,

I'm successfully polling along with `ads.setSPS(ADS1115_DR_860SPS);` using @soligen2010 's fork of the library, but I was wondering if anyone who has been running at a fast SPS with the ADS1115 for a while has any recommendations on if it is a good idea or not. Section 8.1 of the datasheet states there is some internal averaging going on so the faster you sample the noisier the signal gets. Will there be any advantage to manually averaging say 6 readings @ 860 SPS in the Arduino code vs. just one reading @ 128 SPS using the internal hardware averaging?

aka does this bug matter?

thanks for any insights,
Hamish

aviralpatel, 2020-01-12:
Wish I had came across this thread earlier, this could have saved me time and effort to make changes to that library.
btw here is my version of it - https://github.com/aviralpatel/adafruit_ADS11x5/tree/master/Adafruit_ADS1X15-master

aviralpatel, 2020-01-12:
> I have check datasheet, both chips use Bits[7:5], DR[2:0].
>
> ```
> #define ADS1015_REG_CONFIG_DR_MASK      (0x00E0)
> #define ADS1015_REG_CONFIG_DR_128SPS    (0x0000)  // 128 samples per second  // [...]

actually hex codes for 475SPS and 800SPS are wrong 
for 475SPS it is 0x00C0
for 800SPS it is 0x00E0
and  0x00A0 is for 250SPS

caternuson, 2020-02-25:
Closing. Original issue is not an issue, just a quirk of current state of library. There is currently no way to change the data rate with this library.

closed 2020-02-25 by caternuson