adafruit/Adafruit_ADS1X15 #14: SingleEnded read same for all a0-a3 regardless of

npongo, 2016-01-05:
I am have been trying to use the library starting with the samples with an ADS1115. The read single ended ie
 adc0 = ads.readADC_SingleEnded(0);
  adc1 = ads.readADC_SingleEnded(1);
  adc2 = ads.readADC_SingleEnded(2);
  adc3 = ads.readADC_SingleEnded(3);
returns all four values the same regardless of what they are reading (ie ground, vcc or sensor).

soligen2010, 2016-01-05:
They all work for me with a 1115. If you haven't, sure to change the constructor on the example from ADS1015 to ADS1115.

npongo, 2016-01-07:
That was it thanks! And thanks for the library

Benjamin d clark

gordonthree, 2016-01-30:
Similar issue here... channels 0 and 3 read almost  the same, 1 and 2 read almost the same. I'm using the smaller purple breakout instead of the larger blue one. The readings being made are correct, just duplicated... I'm wondering if it's because the chip is gray-market and defective on the insides. I tried grounding unused inputs, as well as tying them to 3.3v - no change.

AIN0: 11577
AIN1: 4639
AIN2: 4639
AIN3: 11553

gordonthree, 2016-01-30:
Additional information; originally used an ESP-01, sda/scl on 0,2 . Also tried a Teensy 3.2. Same readings from both boards.

code I'm using is here: http://pastebin.com/vuciB5EC

Arduino IDE 1.6.7, ESP8266 librares 2.1.0-rc2, not sure on the Teensy version, whatever the latest was as of a week ago.

npongo, 2016-01-30:
My mistake was declaring a ADS1115 as a ADS1015. Once I fixed that is worked great.

gordonthree, 2016-01-30:
Just for kicks, I tried it as an ads1015, same problem.

soligen2010, 2016-02-08:
I don't know what your issue might is, but you can try my fork of the library and see if it is any different. If it turns out to be library software, i would appreciate knowing what it is so i can include the change in my fork.  If you use my fork, it supports additional sample rates so you can change the samples per second to try other read rates.  I have one unit that does not work at 250 SPS, but all the others work. Other features are in this fork you might like too.

[https://github.com/soligen2010/Adafruit_ADS1X15](https://github.com/soligen2010/Adafruit_ADS1X15)

soligen2010, 2016-02-08:
Another thing you can try is increasing the conversion delay, in case you have a chip that is out of spec by being too slow. If you switch to my fork you can call overRideConversionDelay to set the delay (after first setting the SPS).  In the official fork, you would have to edit the .h file.

gordonthree, 2016-02-08:
I am downloading the library right now, will report back shortly!

gordonthree, 2016-02-08:
Dennis, you've cracked the case! These gray-market chips just need more time to take a reading. I set SPS to 32 and overRideConversionDelay to 64, and presto, correct readings assigned to the correct port and inputs tied to ground correctly read 0 now, instead of bouncing around randomly.

  ads.begin();
  ads.setGain(GAIN_ONE); 
  ads.setSPS(ADS1115_DR_32SPS);
  ads.overRideConversionDelay(65);

soligen2010, 2016-02-08:
I'm glad its working!  That delay is double what works for me at 32 sps.  Could you possibly run some tests to find the smallest conversion delay that works for each SPS.  I'd like the library (at least my fork) to work reliably with a broad range of devices, and then people with faster chips can speed things up if they want.

Thanks

soligen2010, 2016-02-09:
I found that I have one of the purple modules in my box that I never used, so I tested it and it is too slow at certain SPS settings.  I don't really like hard coding the conversion delays since there are timing variations, so I went reading through the data sheet and I think I see a way to have the chip say when the conversion is done.  If what I have in mind works, it will take the conversion delay issue off the table because the code will wait until the chip says the conversion is done.  I'll post again when I see if it works.

gordonthree, 2016-02-09:
I really like that idea, as long as polling the GO/DONE bit or whatever it's called doesn't end up eating too many cycles.

I will test speeds on my module for you shortly, haven't had chance to spend time in my lab lately

soligen2010, 2016-02-10:
The new method works! I think no need to test speeds anymore, but please test the new library.  You will need to remove your call to overRideConversionDelay as it now polls once per millisecond.  Eating cycles should not be an issue as it still uses the delay function - just in a loop instead of all at once.

You can grab my new version at 

[https://github.com/soligen2010/Adafruit_ADS1X15](https://github.com/soligen2010/Adafruit_ADS1X15)

soligen2010, 2016-02-10:
It occurs to me that 1 ms delay resolution makes the faster SPS settings rather pointless.

gordonthree, what specifically is your concern of taking too many cycles?  I am considering changing to delay(0) which will tighten the loop and make response faster, but will still do a yield for ESP8266 processing.  Thoughts?

soligen2010, 2016-02-10:
Since this issue is now pertaining it to the soligen2010 fork, and has this thread has deviated from the original issue, I opened a new issue in the soligen2010 fork.  Please continue discussion there.

Link to new issue: [https://github.com/soligen2010/Adafruit_ADS1X15/issues/1](https://github.com/soligen2010/Adafruit_ADS1X15/issues/1)

9H1LO, 2016-02-17:
soligen2010 fork working on esp8266 was getting same readings on all ports using adafruit library

laercionit, 2018-05-11:
Hello, I have the same problem.
I have an ACS721 sensor, all the readings are mixed.

I have the ADS1115 version, everything is correct in code. I use the NodeMCU 12-e

Any suggestions on where the error is?
NOTE: Latest version of LIB: https://github.com/adafruit/Adafruit_ADS1X15

Code:
#include <Wire.h>
#include <Adafruit_ADS1015.h>
Adafruit_ADS1115 ads; / * Use this for the 16-bit version * /

void setup (void)
{
  Serial.begin (115100);
  ads.setGain (GAIN_TWOTHIRDS);
  ads.begin ();
}

void loop (void)
{
  int16_t adc0, adc1, adc2, adc3;

  adc0 = ads.readADC_SingleEnded (0);
  adc1 = ads.readADC_SingleEnded (1);
  adc2 = ads.readADC_SingleEnded (2);
  adc3 = ads.readADC_SingleEnded (3);
  Serial.print ("AIN0:"); Serial.println (adc0);
  Serial.print ("AIN1:"); Serial.println (adc1);
  Serial.print ("AIN2:"); Serial.println (adc2);
  Serial.print ("AIN3:"); Serial.println (adc3);
  Serial.println ("");
  delay (1000);
}

Serial Output:
AIN0: 11397
AIN1: 11055
AIN2: 13487
AIN3: 5182

laercionit, 2018-05-14:
When the port has no sensor connected, should it return a fixed value or will it vary?
I will carry out the test with your lib.

This is the link below: https://github.com/soligen2010/Adafruit_ADS1X15

For your fork do I need to make some specific adjustments to ensure that the channels have the readings independently?

soligen2010, 2018-05-14:
It will vary unless you wire in a pull up or down resistor.  My fork should not need adjustment.  It checks the chip's status instead of using a fixed wait time.

laercionit, 2018-05-14:
Hello @ soligen2010

I need to read two sensors, a voltage sensor (zmpt101b) and an amperage sensor (acs712)

Can I connect to any port or recommend using a specific port on ADS1115?

About pulldown or pullup, if using this would not affect the reading of the analog sensors?

soligen2010, 2018-07-09:
Pull up or down may affect the readings depending on the impedance of the sensor output.  If you are connecting a port to a sensor, then don't use pull up/down.  The comment about pull up/down was only for a port with no sensor connected to avoid random fluctuations.

salami738, 2018-10-30:
Thanks @soligen2010. Your library works flawlessly with the purple (clone) boards.

MkLHX, 2019-06-07:
Hi there! 
Same problem here with ESP32.
 i using the ADS1115 instance and this test sketch

`#include <Wire.h>
#include <Adafruit_ADS1015.h>

Adafruit_ADS1115 ads;  /* Use this for the 16-bit version */
//Adafruit_ADS1015 ads;     /* Use thi for the 12-bit version */

void setup(void) 
{
  Serial.begin(115200);
  Serial.println("Hello!");
  
  Serial.println("Getting single-ended readings from AIN0..3");
  Serial.println("ADC Range: +/- 6.144V (1 bit = 3mV/ADS1015, 0.1875mV/ADS1115)");
  
  // The ADC input range (or gain) can be changed via the following
  // functions, but be careful never to exceed VDD +0.3V max, or to
  // exceed the upper and lower limits if you adjust the input range!
  // Setting these values incorrectly may destroy your ADC!
  //                                                                ADS1015  ADS1115
  //                                                                -------  -------
  // ads.setGain(GAIN_TWOTHIRDS);  // 2/3x gain +/- 6.144V  1 bit = 3mV      0.1875mV (default)
  // ads.setGain(GAIN_ONE);        // 1x gain   +/- 4.096V  1 bit = 2mV      0.125mV
  // ads.setGain(GAIN_TWO);        // 2x gain   +/- 2.048V  1 bit = 1mV      0.0625mV
  // ads.setGain(GAIN_FOUR);       // 4x gain   +/- 1.024V  1 bit = 0.5mV    0.03125mV
  // ads.setGain(GAIN_EIGHT);      // 8x gain   +/- 0.512V  1 bit = 0.25mV   0.015625mV
  // ads.setGain(GAIN_SIXTEEN);    // 16x gain  +/- 0.256V  1 bit = 0.125mV  0.0078125mV
  
  ads.setGain(GAIN_ONE);
  ads.begin();
  
}

void loop(void) 
{
  int16_t adc0, adc1, adc2, adc3;

  adc0 = ads.readADC_SingleEnded(0);
  adc1 = ads.readADC_SingleEnded(1);
  adc2 = ads.readADC_SingleEnded(2);
  adc3 = ads.readADC_SingleEnded(3);
  Serial.print("AIN0: "); Serial.println(adc0);
  Serial.print("AIN1: "); Serial.println(adc1);
  Serial.print("AIN2: "); Serial.println(adc2);
  Serial.print("AIN3: "); Serial.println(adc3);
  Serial.println(" ");
  
  delay(1000);
}`

i trying to read EC sensor(A0) and pH sensor(A1) simultaneously, i have some trouble

when using the test sketch, 
A0 = on the first loop execution 819 an then 2
A1 =10906
A2 = 2
A3 = 2

when i read only one ADC input i have the correct value on each input.

So, @tdicola there is a fix to fix it?

MkLHX, 2019-06-07:
Hi find solution from https://forum.arduino.cc/index.php?topic=561546.0
just up the conversion delay : #define ADS1115_CONVERSIONDELAY         (8)
in my sketch i add 
```
#ifdef ADS1115_CONVERSIONDELAY         
#undef ADS1115_CONVERSIONDELAY
#define ADS1115_CONVERSIONDELAY 10
#endif
```

DaveCalaway, 2019-06-18:
I'm on ESP32 + ADS1015 and i have an interesting problem.
A0 doesn't read anything.
A1 and A2 works ok.
A3 works also ok, but when i measure the A3 value, A0 give me the same value!

MkLHX, 2019-06-18:
Finaly, i forked the repo and change CONVERTION_DELAY from (8) to (10) and i use my own version.

DaveCalaway, 2019-06-18:
> Finaly, i forked the repo and change CONVERTION_DELAY from (8) to (10) and i use my own version.

Thank you so much for your job. 
Works grate with Vin (aka 5v), on only with 3v3.

I have edited the code like you wrote in this pull: https://tinyurl.com/y2a5rj55

But if i try to load those lines in my code, they doesn't work.
```
#ifdef ADS1115_CONVERSIONDELAY    
#undef ADS1115_CONVERSIONDELAY
#define ADS1115_CONVERSIONDELAY 10
#endif
```
It's strange because the preprocessor directives looks good....

MkLHX, 2019-06-18:
Yep not sure of this workaround, so i fork and modify the lib here : https://github.com/GreenPonik/Adafruit_ADS1X15

ipsod, 2019-10-17:
@DaveCalaway I've got that same problem you described, with A0 not taking readings, but following A3's readings.  Did you ever solve it?

DaveCalaway, 2019-10-17:
> @DaveCalaway I've got that same problem you described, with A0 not taking readings, but following A3's readings. Did you ever solve it?
Simple uses the MkLHX library instead the adafruit library: https://github.com/GreenPonik/Adafruit_ADS1X15

ipsod, 2019-10-17:
Works for me, thanks!

Jobui21, 2020-09-08:
> Finalmente, bifurqué el repositorio y cambié ADS1115_CONVERSIONDELAY de (8) a (10) y uso mi propia versión.

@MkLHX    Hola, probé su libreria pero solo sigo con el mismo problema, A0 no me da una respuesta pero A1,A2 y A3 si me trabajan bien, me podrían colaborar, por favor

MkLHX, 2020-09-09:
Could you try with an other ADC ? don't know why A0 don't works and others works. maybe you have a hardware issue.