adafruit/Adafruit_ADS1X15 #15: New Fork with consolidated pull requests and other enhacements

soligen2010, 2016-01-18:
Please try out the soligen2010 fork.

Since pull requests don't seem to be being picked up, I took several pull requests from other people, my own enhancements, and have created my own fork. It should be backward compatible with code using the old library.

I have tested in Arduino nano/uno and ESP8266 with ADS1115.  I don't have an ADS1015 to test with, so if someone can test with it, I would appreciate it.  Especially test each Samples per Second setting to validate the the conversion delay works in each case, and the _V methods with each gain setting to validate the proper volts per bit are being used.

My hope is that with all the changes consolidated, perhaps the owners of this library will adopt this version, as it has a lot of nice things added.

soligen2010, 2016-02-10:
Here is a link to my fork.  I want to encourage people to use this fork because I think it has a lot of things that are the direction this library should go.  I have opened up the issues list for my fork as well so any issues there don't get confused with the original library.

[https://github.com/soligen2010/Adafruit_ADS1X15](https://github.com/soligen2010/Adafruit_ADS1X15)

madvoid, 2016-08-21:
@soligen2010 Thank you for your fork of this library! I had a problem where the ADS1115 worked on an Arduino Mega2560 but not on a SAMD21 based board, as described [here](https://forums.adafruit.com/viewtopic.php?f=19&t=101797&sid=0455ea88e14c21f2a72333c6f9ae391a). Your library fixed the problem and I'm assuming it was because of sampling time differences. I'm leaving this comment here in case anyone has similar troubles as me and is looking for a solution.

soligen2010, 2016-08-23:
You are welcome :)

caternuson, 2021-03-25:
Hopefully most of this has been covered with recent updates. @soligen2010 Thanks again for your efforts here.

closed 2021-03-25 by caternuson