adafruit/Adafruit_ADS1X15 #18: Incorrect gain comments

madvoid, 2016-08-04:
This is not a bug, but what I think is a mistake in the comments. However, since people may design differently because of the comments, I thought I should bring it to attention. In the example code, there is comment that discusses the danger of incorrectly setting the gain on the ADC ([here](https://github.com/adafruit/Adafruit_ADS1X15/blob/master/examples/singleended/singleended.pde#L15) is an example from `singleended.pde`, but the comment is in all of the examples). 

It says to be careful of setting the gain because a signal that exceeds the upper or lower limits of the gain can destroy the device. However, I don't think that's correct. According to [this link](https://e2e.ti.com/support/data_converters/precision_data_converters/f/73/t/378122) and [this link](https://e2e.ti.com/support/data_converters/precision_data_converters/f/73/p/398187/1407689#1407689) from the TI forums, a signal that is greater than the PGA won't damage the ADC, only a signal that is over VDD+0.3V or under GND-0.3V can do that. A signal that exceeds the upper or lower limits of the gain will simply give a 0 or full scale reading.

Anyways, just thought I'd point it out. Thanks!

soligen2010, 2016-08-09:
In the data sheet absolute maximums it says vdd_+ .3, but the electrical
characteristics section that says the input voltage range is ±4.096/PGA so
with a 2/3 pga input voltage can get to 6.144.  To be truely safe, best to
adhere to both limitations

You may choose to run voltage higher than ±4.096/PGA and have no issues,
but technically it is outside the parameters of the datasheet and therefore
the behaviors are not defined.  I think it is better to leave the comments
in the examples conservative because a lot of people don't read the data
sheets. Regardless of who was answering on the forum, I think it is best
for the comments to be based only off of the official data sheet.  People
like you who are willing to dig a little deeper can decide for themselves
what to do.

madvoid, 2016-08-09:
Fair enough, I'll mark this as closed. Thank you!

closed 2016-08-09 by madvoid