adafruit/Adafruit_ADS1X15 #21: Why is there no continuous sampling mode (outside the comparitor) and are the rates correctly specified?

drp0, 2016-10-31:
Is this a limitation of the chip-
I can not see any restriction in the chip manual that suggests this.
It would be helpful to have a call to achieve continuous sampling on one port.
Preferably with polling of the device to release the adc value as soon as the conversion is complete.

I note that the delay for conversion on the ADS115 appears to be set to 8ms
  According to the manual, the default conversion rate is 128 sps-
  All the conversion rates are miss named/ over stated?
  Certainly bits 7:5 set to 100b would give 128sps according to page 19 of the manual:
    Bits [7:5] DR[2:0]: Data rate
    000 : 8SPS
    001 : 16SPS
    010 : 32SPS
    011 : 64SPS    
    100 : 128SPS (default)    
    101 : 250SPS
    110 : 475SPS
    111: 860SPS
128sps takes 7ms to convert so an 8mS delay sensibly errs 1mS on the safe side and confirms my suspicion of the actual data rates.
 
David

soligen2010, 2016-10-31:
You may want to check out my fork.  I have no fixed delay, so it is much more responsive at higher data rates.

Also, the only reason that there is no continuous sampling is simply because no-one has written the method yet.  I think you could do it in my fork simply by setting the comparitor to a value that never triggers (but I have never tested this).  Adafruit doesn't seem to take pull requests, but I would take one into my fork if you want to explicitly add the functionality.  Please post back how it goes.

drp0, 2016-11-05:
@soligen2010 
I dowloaded the code at [https://github.com/soligen2010/Adafruit_ADS1X15](url). I presume this is the fork you referred to. It appears more developed than the default adafruit library.
I particularly appreciated the use of polling to wait for the conversion to complete.
Do you agree with the SPS values I posted above?

I tested a new ADS1115 which worked briefly then ceased. It will no longer respond to Wire.endTransmission(); from a simple setup:

```C++
Wire.beginTransmission(0x48);
Wire.write((uint8_t)0x01); //config
Wire.endTransmission();
```
So the chip must be dead. How reliable have you found them to be?

David

drp0, 2016-11-05:
I now have a working chip.
I can confirm the sps values listed in the ads1115 manual [https://cdn-shop.adafruit.com/datasheets/ads1115.pdf](url) page 19.

Some code to test continuous mode verses  the library single shot mode:
I attached a wire between pin 3 on a mega and the ads1115 Alert/Rdy line to determine, by interrupt, when a conversion has just completed. 
Any irq attachable pin would do.

``` C++
#include <Wire.h>
#include <Adafruit_ADS1015.h>

Adafruit_ADS1115 ads(0x48);   /* Use this for the 16-bit version */
//Adafruit_ADS1015 ads; /* Use thi for the 12-bit version */

float sixteenbit = 32768;
float gainone = 4.096/sixteenbit;
float gaintwothirds = gainone * 3 / 2; // in volts
float gain = 0.1875F;
byte address = 0x48;
float vpb; // returned by ads1015 library
volatile boolean waiting = true;

void setup(void) {
Serial.begin(115200);
  while(!Serial) yield();
Serial.println("ADS1115\n");

Serial.println("Getting single-ended readings from AIN0");
Serial.println("ADC Range: 0 to 6.144V (1 bit = 0.1875mV)");

vpb = ads.voltsPerBit();
Serial.print("Default library volts per bit "); Serial.println(vpb, 8);
Serial.println();
Wire.begin(); continuous();Wire.end();

ads.begin();
single();
}

void loop(void) {
yield();
}

void single(){
Serial.println("\nSingle shot sampling A0");
Serial.print("Entry "); readconfig();
Serial.println();
Serial.flush();

byte maxbytes = 10;
unsigned int retvals[maxbytes];
unsigned long average = 0;
byte i;

unsigned long startTime = micros();
  for (i = 0; i < 10; i++) {
  retvals[i] = ads.readADC_SingleEnded(0);
  }
unsigned long endTime = micros();

float elapsed = endTime  - startTime;
float sps = maxbytes * 1000000.0 / elapsed;
elapsed = elapsed / (maxbytes * 1000.0); // mS
Serial.print("Time per sample "); Serial.print(elapsed, 6);
Serial.print(" mS, sps "); Serial.println(sps);
  for(i = 0; i < maxbytes; i++){
  average += retvals[i];
  Serial.print("AIN0: "); Serial.print(retvals[i]);
  float mv = gain * retvals[i];
  float v = mv / 1000;
[... 96 lines cut ...]
Wire.endTransmission();
Serial.print("Exit: "); readconfig();
Serial.flush();
}

void readconfig(){
unsigned int retval;
Wire.beginTransmission(address);
Wire.write((uint8_t)0x01); //config
Wire.endTransmission();
  Wire.requestFrom(address, (uint8_t)2);
  retval = (Wire.read() << 8) | Wire.read();
Serial.print("Config Register "); Serial.println(retval, BIN);
}

boolean hasconverted(){ // single shot conversion detection
Wire.beginTransmission(address);
Wire.write((uint8_t)0x01); //config
Wire.endTransmission();
Wire.requestFrom(address, (uint8_t)2);
unsigned int myconfig = ( (Wire.read() << 8) | Wire.read() ) & 0x8000;
  if (myconfig = 0) return false; else return true;
}
```
David

soligen2010, 2016-11-07:
As for chip reliabaility, I had one go bad.  One SPS stopped working (the default SPS) but the others seemed to still work.

As for timing, I have seen variance in conversion times.  The purple eBay modules seem to be slower than the blue modules.  Out-of-spec slow, which is why I put in the polling technique.  I have not had any issues with these cheap modules after this change.  I would think interrupt driven code would also be resilient to these slow chips - you just may see more time than expected between interrupts.

Do you plan to modify the library and put in a pull request against my fork?

andras-kiss, 2016-11-07:
I get a sample in every 3-4 ms from the purple modules (~300 sps). Is it possible to get a higher rate with your polling technique?

soligen2010, 2016-11-07:
I think Chinese knockoff chips may vary.  Please try it ans see.  You may also want to check that the values are not repeating.  Was a year ago I tested this stuff, and I am going by memory, but I think if your interval is too small you simply get the same conversion result a second time, so the code may appear to work with a given fixed interval, but you don't necessarily get the same value.  The polling in theory should give you a result in the shortest time possible, plus the polling overhead of course.

Here is a link to the discussion on the whole conversion delay issue.  The conversation starts here, then continues as an issue under my fork (like to the continuation is in the last post)

[https://github.com/adafruit/Adafruit_ADS1X15/issues/14](url)

andras-kiss, 2016-11-07:
How do I know that they are duplicates of the same conversion instead of just different conversions that resulted the same value? This is an output form the ads1115:

time (s), voltage (mV)
3.4890 334.9531
3.4920 334.9375
3.4960 334.9688
3.5000 335.0000
3.5030 335.0000
3.5060 334.9688
3.5100 334.9531
3.5130 334.9375
3.5160 334.9844
3.5190 334.9844
3.5230 335.0000
3.5260 334.9844
3.5290 334.9375
3.5320 334.9375
3.5360 334.9688
3.5390 334.9844
3.5430 334.9844
3.5460 334.9688
3.5500 334.9531
3.5530 334.9375
3.5560 334.9688
3.5590 335.0000
3.5630 335.0000
3.5660 334.9844
3.5690 334.9531
3.5720 334.9375
3.5760 334.9688
3.5790 335.0000
3.5820 335.0000
3.5860 334.9531
3.5900 334.9375
3.5930 334.9375

The time interval is sometimes 3, sometimes 4 ms. This is the code I used to obtain it:

> #include <Wire.h>
> #include <I2Cdev.h>
> #include <ADS1115.h> // from https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/ADS1115/ADS1115.h
> 
> 
> ADS1115 adc0(ADS1115_ADDRESS_ADDR_GND); 
> float scalefactor = 0.015625; // change this to match your amplifier settings
> float millivolts = 0.0; // The result of applying the scale factor to the raw value
> float time;
> 
> int16_t start;
> int16_t elapsed;
> uint16_t currentADCreadings;  
> 
> void setup(void)
> {
>   Serial.begin(115200);  
>   Wire.begin();
>   adc0.initialize(); // initialize ADS1115 16 bit A/D chip
>   //Serial.println(adc0.testConnection() ? "ADS1115 connection successful" : "ADS1115 connection failed");  
> 
>   // To get output from this method, you'll need to turn on the 
>   //#define ADS1115_SERIAL_DEBUG // in the ADS1115.h file
>   adc0.showConfigRegister();
>   adc0.setRate(ADS1115_RATE_860);  //860 samples/sec
>   adc0.setMode(ADS1115_MODE_CONTINUOUS);  //continuous sampling
>   adc0.setGain(ADS1115_PGA_0P512);  // 8x gain   +/- 0.512V  1 bit = 0.015625mV  710kΩ
>   adc0.setMultiplexer(ADS1115_MUX_P0_N1);   // sets mux to differential
> }
> 
> void loop(void)
> {  
>   time = 0.001 * millis();
>   currentADCreadings=adc0.getConversion(); //it takes 1.16 ms for the ADC to do a new conversion
>   Serial.print(time,4);
>   Serial.print(' ');
>   Serial.println(adc0.getMilliVolts(),4);
>   //delay(1000);
> }

There was an arduino code I used to measure the frequency, and that showed around 900 sps. Unfortunately I can't find or remember it. It had some sort of trigger and then did 1000 conversions, and measured the elapsed time.

soligen2010, 2016-11-07:
In the testing (see other thread) each read was from a different channel but gave the same result - given the circuit this was impossible, and the timing change fixed it.

With yours using only one channel, it depends on how low noisy your circuit is.  I cant assess if yours is dups or not.  I also don't know what the fork of the library you are using is doing.  I suggest trying my library that polls and see if there is a difference,

soligen2010, 2016-11-07:
Looks like you are doing continuous conversions, which my lib currently doesn't do.  Just realize that unless there is a long enough delay built into getConversion() that you will get the results of the last completed conversion until the new conversion completes.  I would think using the alert/ready pin as drp0 is trying to do would be a much better way to use continuous mode rather than using delays and/or polling.

soligen2010, 2016-11-08:
I finally got a change to review things in the data sheet.  I found that adding a method to set up continuous conversion was pretty easy, so I added it.  Just for grins I also added a method for the window comparator.

startContinuous_SingleEnded(uint8_t channel);
startWindowComparator_SingleEnded(uint8_t channel, int16_t lowThreshold, int16_t highThreshold);

Please test and let me know if there are any issues.  I didn't put the interrupt handling in the library, I think that is best left in the sketch.

This is in the fork soligen2010/Adafruit_ADS1X15

drp0, 2016-11-08:
The alert pin works well.
I have refined my test code to include an irq based comparator.
I have no difficulty in getting the stated sps up to and including the top rate.
Two points to note: 
If the input is ground, the ADS115 leaves the alert line low- there is no falling pulse to detect.
A simple check on the arduino alert pin solves this.
When grounded, some of the returned values are minutely negative. (Even though it is a single ended setup)
For this reason the returned value storage array has to be int, not unsigned int

I have also added a precautionary time out on the wait for the pulse in case the line remains high.
The sketch includes support for an I2C, 4 line lcd panel.
````C++
// ads1115_singleEnd2.ino
// D.R.Patterson
// 06/11/2016

// nb when the input is zero the ready/alert line stays low

#include <LiquidCrystal_I2C.h>
// set the LCD address to 0x27 for a 20 chars 4 line display
// Set the pins on the I2C chip used for LCD connections:
//                    addr, en,rw,rs,d4,d5,d6,d7,bl,blpol
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);

#include <Wire.h>
#include <Adafruit_ADS1015.h>

// demonstratable values for gain multipliers
unsigned int sixteenbit = 32768;
float gainone = 4.096 / sixteenbit;
float gaintwothirds = gainone * 3 / 2; // volts

volatile boolean waiting = true; // irq flag that the ready line has gone low
uint16_t myconfig; // store config flags
byte i;
unsigned int spsMask = 0x0080; // bit mask for the sps value
unsigned int gainMask = 0x0000; // bit mask for gain
float gain = 0.0001875F; // the multipler for 2/3rd gain in mV
unsigned int libGain;
unsigned int libSps;
unsigned long twait;
#define MUX_DIFF_0_1 (0x0000)  // Differential P = AIN0, N = AIN1
#define MUX_DIFF_0_3 (0x1000)  // Differential P = AIN0, N = AIN3
#define MUX_DIFF_1_3 (0x2000)  // Differential P = AIN1, N = AIN3
#define MUX_DIFF_2_3 (0x3000)  // Differential P = AIN2, N = AIN3

// edit these first ****************************************************

unsigned int sps = 128; // sps 8, 16, 32, 64, 128, 250, 475, 860

// Do not exceed VDD + 0.3V, or the gain set max!
//                                                       ADS1015          ADS1115
//                                                       -------------    -----------------
// ads.setGain(GAIN_TWOTHIRDS); // 2/3x gain  +/- 6.144V  1 bit = 3mV     0.1875mV (default)
// ads.setGain(GAIN_ONE);       // 1x gain    +/- 4.096V  1 bit = 2mV     0.125mV
// ads.setGain(GAIN_TWO);       // 2x gain    +/- 2.048V  1 bit = 1mV     0.0625mV
// ads.setGain(GAIN_FOUR);      // 4x gain    +/- 1.024V  1 bit = 0.5mV   0.03125mV
// ads.setGain(GAIN_EIGHT);     // 8x gain    +/- 0.512V  1 bit = 0.25mV  0.015625mV
// ads.setGain(GAIN_SIXTEEN);   // 16x gain   +/- 0.256V  1 bit = 0.125mV 0.0078125mV

byte setgain = 0; // 0 = 2/3 rds gain, 1, 2, 4, 8, 16

const byte maxbytes = 10; // number of test values

byte address = 0x48; // ads1115 address (requires address line to ground)
//byte address = 0x49; // address line to VDD
//byte address = 0x4A; // address line to SDA
//byte address = 0x4B; // address line to SCL

const byte alert = 3; // pin must be able to support irq
[... 487 lines cut ...]
boolean hasNotConverted(){ // single shot conversion detection
Wire.beginTransmission(address);
Wire.write((uint8_t)0x01); //config
Wire.endTransmission();
Wire.requestFrom(address, (uint8_t)2);
byte retval = Wire.read() & 0x80; // examine high bit
Wire.read();
  if (retval == 0) return true; else return false;
}

String pad(int temp, byte no){
String todo = String(temp);
byte L = todo.length();
  for(byte j = 1; j < no - L; j++) todo = " " + todo;
return todo; 
}

unsigned int chkmask(byte adc){
  if (adc == 0) return 0x4000;
  if (adc == 1) return 0x5000;
  if (adc == 2) return 0x6000;
return 0x7000;  
}
````
David

soligen2010, 2016-11-08:
You Wrote
      If the input is ground, the ADS115 leaves the alert line low- there is no falling pulse to detect.
      A simple check on the arduino alert pin solves this.

Not sure I understand this.  BTW the data sheet does say that the alert pin needs a pull up resistor.  Setting the pin to INPUT_PULLUP is probably sufficient.

You Wrote
     When grounded, some of the returned values are minutely negative. (Even though it is a single ended
     setup)

Yes I noticed this too.  The original adafruit library had a bug because it didn't use a signed return function.  

What I do in the sketch is if the returned value is < 0 set it to 0.  After that check you can store it in an unsigned variable if you want.  Another option is to software calibrate zero by adding a constant to the result to move the lowest possible value up to zero, but if your code is sensitive to negatives, I would still include the <0 check.

drp0, 2016-11-08:
It is difficult to see, but it looks like the board already has a pullup resistor on the alert line-
I tried with just input and input pullup, there was no difference.
I tried attaching a positive(rising) pulse irq to the alert line.
For ground input, the irq never fired.
I appreciate your observations on ground accuracy. 
The minute negative voltage returned is in the 3rd or 4th decimal place.
I can live with that.
I would be interested in others trying my test sketch to confirm sps on other setups.
David

soligen2010, 2016-11-08:
I don't currently have a circuit handy to test on.  None of my projects have used the alert pin.  I'll post pack when I get a chance to put together a test setup.

soligen2010, 2016-11-29:
I finally got to make up a circuit to test continuous.  I am not seeing the issue with the alert pin staying low.  It is triggering fine and causing interrupts as expected.  I did see an issue where the first interrupts returned a bogus conversion reading, so I added a non-differential read to the startContinuous function to prime the conversion register with a valid reading.  Also added support for differential continuous mode.

I put a continuous example in with the new version of the library.

drp0, 2016-11-29:
A second (expensive) breakout board chip failed. So I bought 2 cheap breakout boards on e-bay for less than the price of the adafruit version.
The e-bay chips have not failed and do not keep the alert pin low-
different internal programming!
Re the bogus 1st reading, I believe there is reference to dis-regarding the first reading in some versions of the chip manual, under certain setup conditions.
The arduino chip manual advises the same for internal adc readings when setup values are changed in the registers.
It is good practice to ignore at least one, if not 2, first values for a lot of devices.

soligen2010, 2016-12-02:
I'm glad its working for you now. Odd to have the adafruit stuff fail.  I would have expected it to be higher quality.  All my testing was with e-bay modules.

For future people reading this thread, the continuous mode support (and other good stuff) is in the soligen2010 fork at [https://github.com/soligen2010/Adafruit_ADS1X15](https://github.com/soligen2010/Adafruit_ADS1X15)

closed 2017-06-09 by drp0

misterjoa, 2019-10-04:
@soligen2010 thank you very much, your fork solves all the problems that I had with the lib !!!