adafruit/Adafruit_ADS1X15 #23: ADS1015 conversion ready timing

davidegironi, 2017-04-24:
Hello and thank you for this library.
I'm testing this over a ADS1015 IC. I've notice that the single-ended example only works if a 7ms delay is added.
I suppose, but I'm not sure cause I've not investigated further, that the above macro have to be inverted
```
    #define ADS1015_CONVERSIONDELAY         (1)
    #define ADS1115_CONVERSIONDELAY         (8)
```
Hope this helps.

soligen2010, 2017-04-24:
This version of the library is old.  Adafruit doesn't seem interested in taking pull reuests to improve the library.  I suggest you check out the soligen2010 fork as an alternative where the whole conversiondelay approach has been removed

davidegironi, 2017-04-24:
Thank you for your suggestion, I will check it out!

caternuson, 2020-02-26:
With the way the library currently works, the ADS1015 will run at 1600 samples per second. So each sample shouldn't take more than 1 / 1600 = 0.000625 ~= 0.6 ms. Rounding to 1ms *should* be enough. Needing 7ms seems like too much.

The ADS1115 runs slower, at 128 samples per second. 1 / 128 = 0.0078125 ~= 7.8ms, which is where the other delay value comes from.

Closing for now. If this is still an issue, please provide more info about your setup and an example that demonstrate the issue.

closed 2020-02-26 by caternuson