adafruit/Adafruit_ADS1X15 #31: ESP32 support

choopk, 2018-04-12:
Hi is this library compatible with esp32?

chinswain, 2018-07-10:
Yes

Bzzz, 2019-05-29:
Only compiles when not touching gain settings.

If I include `ads.setGain(gainsetting);` (which works with arduinos), compile fails:

```
/libraries/Adafruit_ADS1X15-master/Adafruit_ADS1015.h:137:13: note:   initializing argument 1 of 'void Adafruit_ADS1015::setGain(adsGain_t)'
   void      setGain(adsGain_t gain);
             ^
exit status 1
invalid conversion from 'long int' to 'adsGain_t' [-fpermissive]
```

Currently looking into what's wrong, it's probably some issue with the enum block of possible gain settings that does not play well with the ESP. Doesn't matter if I pass a variable to setGain() or one of the constants like ADS1015_REG_CONFIG_PGA_0_256V directly, so I don't think it's an ordinary conversion problem. Not a great C coder, though.

rel1ct, 2019-09-12:
For ESP32 in file Adafruit_ADS1015.h
#define ADS1115_CONVERSIONDELAY         (9)  // default 8