adafruit/Adafruit_ADS1X15 #32: samd21 timing

ladyada, 2018-07-01:
https://forums.adafruit.com/viewtopic.php?f=19&t=101797&p=681434#p681434

soligen2010, 2018-07-09:
My fork already has a timing loop, which to my knowledge has resolved all the timing issues for people who have tried it.  It also has many other enhancements and supports more features of the this chip than this original version of the library.

Ladyada, I originally put in pull requests for the changes I made, but nothing ever happened with them.  There have been a number of other people who over the past couple years who had problems that my fork resolved.  I have to wonder how many people have had issues go unresolved because they never found my fork. Might I suggest you look at my fork, and if you like it, simply adopt it as a new version of this library.  A lot of people come here for a library supporting the chip but never find my fork.

ladyada, 2018-07-09:
hi @soligen2010 do you mean this PR?
https://github.com/adafruit/Adafruit_ADS1X15/pull/16
a full rewrite makes it a lot harder to test - which is why we didnt merge it :)

soligen2010, 2018-07-09:
No doubt testing is a significant effort. This PR is really an accumulation of many changes/additions over time, not a full rewrite.  I was careful to keep the library backwards compatible with the original, so any testing setups you have for the original should still work just fine against the fork.

I also included the sketch I use for testing in the samples, just need to feed the inputs via voltage dividers as described and it will spin through many use cases.

It's there for you if you want it.  It has no known bugs for some time now, and it has helped a number of your customers.

ladyada, 2018-07-09:
ok can you squash the commits? we'll take a look!

soligen2010, 2018-07-09:
Squashing seems like a good idea.  I have not done this before and a quick google tells me this is a bit more involved than just pressing a button.  As I am traveling part of this week, it will likely be next week before I have time to figure this out.  

Looks like I have to use the GIT command line interface, which I have very little experience with.

ladyada, 2018-07-09:
@caternuson in the mean time can you test the new library/fork by @soligen2010

soligen2010, 2018-07-09:
I read some more on squashing.  Since PR #16 has no merge conflicts, looks like the easiest thing  recommend  would be to do a "Squash and Merge" instead of just a Merge.  

Before merging, to see the net change of all the commits combined, all you need to do is click on the "files changed" link on the pull request page instead of going into each commit.

If you are OK with this, I think it will be easiest/safest, as opposed to me having to run a bunch of commands that I don't fully understand to do the squash and then update the pull request.  Please let me know.

ladyada, 2018-07-09:
ok sure - once @caternuson has a chance to test it, we can look at the merge!

caternuson, 2018-07-10:
@ladyada Tested all examples with UNO+ADS1015 and Itsy M0+ADS1115 and verified working.

@soligen2010 Nice work. Instead of doing an actual code review on the PR, I'll just summarize here:
* change all examples from .pde to .ino
* change all examples to use ADS1015 by default
* change all examples to use 9600 baud for serial
* singleended example still has comment typo "Use thi"
* continuous example does double config of `alertPin` in `setup`
* remove test_ADS1115 from examples (a little too specialized)
* why library.json? (platformio?)
* library.properties - keep unchanged, we'll tweak this after PR merge when we make a matching release in our repo to trigger arduino lib man stuff

soligen2010, 2019-01-21:
I made what changes I thought made sense.  

1) I did not change .pde to .ino.  The original library has .pde, so I didn't change them and not sure if a rename will merge.  I suggest you re-name them after the merge.

2) I left test_ADS1115 in.  I don't know how to remove just this one thing from the stream of changes,  Please delete it after the merge if you don't want it. I personally think it is of value to anyone modding the library, so I suggest you keep it.

3) Library.properties.  I didn't see any different other than a blank line, which I removed

jonra1993, 2020-07-06:
Hi @soligen2010 I am communicating ADS1015 with Arduino nao but communication takes too much time some ms. Did you have similar issues in the past?

caternuson, 2021-03-25:
Hopefully fixed with #59. Can reopen if there's still an issue.

closed 2021-03-25 by caternuson