adafruit/Adafruit_ADS1X15 #33: Please fix Adafruit_ADS1015.h

ppedro74, 2018-09-10:
Please add the code bellow to Adafruit_ADS1015.h:

#ifndef _LIB_ADAFRUIT_ADS1015_
#define _LIB_ADAFRUIT_ADS1015_

....

#endif

To avoid symbols redefinition errors.

caternuson, 2020-02-26:
fixed by #47

closed 2020-02-26 by caternuson

HansJoachimKrahe, 2021-05-13:
got error: 

E:\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':

E:\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:133:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'

     _spi->transfer(buffer, len);

                               ^

is it an incompatibility between Adafruit_1X15 V 2.1.1  and Adafruit_BusIO  V 1.7.3  ????