adafruit/Adafruit_ADS1X15 #34: i2c pins

ramarro123, 2018-09-20:
hi,

can i suggest to add i2c pins to begin constructor? in that way, non standard controller are able to use the library without need to modify it.

some esp8266 are limited and have to reuse pins for i2c, so using Wire.begin(x,y) can be quite useful.

YordanYanakiev, 2018-11-18:
I am just in the same boat.
Trying to use 2 x ADS1115 with ESP32, but stuck on the lack of constructor with pins params..

ladyada, 2018-11-19:
hiya we wont' be adding this - the board definition for your ESP32 board should have th default Wire pins defined :)

closed 2018-11-19 by ladyada

YordanYanakiev, 2018-11-19:
ESP32 boards have 2 x I2C wires.
Not adding this automatically kills the possibility of using the second I2C wire with a second ADS1115.

YordanYanakiev, 2018-11-19:
I've replaced everywhere Wire with Wire1. 
as well did this :
	gpio_pad_select_gpio( pinSDA );
	gpio_set_direction( pinSDA, GPIO_MODE_INPUT_OUTPUT );

	gpio_pad_select_gpio( pinSCL );
	gpio_set_direction( pinSCL, GPIO_MODE_INPUT_OUTPUT );

inside the constructor. no luck yet. :'(

YordanYanakiev, 2018-11-20:
This is the version of the library with my modification ( RX/TX -> SDA/SCL on second I2C ) which (maybe) works ( at least it looks like on first sign ).
I would be thankful if someone check it for optimization purposes.

[ADS1115EX.zip](https://github.com/adafruit/Adafruit_ADS1X15/files/2597507/ADS1115EX.zip)