adafruit/Adafruit_ADS1X15 #36: uint16_t instead int16_t for readADC_SingleEnded

AdrianBan, 2019-01-05:
Hi,

I've used this library for my ADS1115 ADC.
After playing a little bit with the library and looked into the code I've notice the function readADC_SingleEnded returns an uint16_t instead int16_t.

Checking the differential function which is returning an int16_t and checking the the documentation of the IC both readings returns -/+ values.
There is a reason why SingleEnd reading is returning uint16_t?

Kind regards,
Adrian

caternuson, 2020-02-25:
The ADS1115 does not allow negative voltage inputs w.r.t GND on any input pin. So in a single ended configuration, there is no way to produce a negative result.

closed 2020-02-25 by caternuson

broadstone, 2020-10-30:
Are you sure about this? The mux just connects the negative side to GND pin. This does not prevent negative numbers, just makes them incredibly unlikely. The datasheet says nothing about preventing AIN1-3 from seeing -0.1V relative to ground, especially if a hobbyist uses a bunch of long wires and makes themselves an antenna by accident.

caternuson, 2020-10-30:
[image]

broadstone, 2020-11-01:
Yes, that means you will very likely damage something if it is exposed to <-0.3V relative to GND.  But all it will take is something like a -.001V to produce at 5-6 counts into the negative even at 5.5V supply, producing a surprisingly large number when presumed as unsigned.  This can lead to rare unexpected outcomes.