adafruit/Adafruit_ADS1X15 #37: how to use more than one ads1115 on a single i2c?

satdj, 2019-01-10:
Hi

how can i use more then one ads1115 at the same i2c and same time?

i can switch hardware Addr. at ads1115 (addr to gnd, vdd, sda or sdc) itself, but how at the libary?

satdj, 2019-01-10:
this is only for one ads1115! 

defined at 
https://github.com/adafruit/Adafruit_ADS1X15/blob/c559bafad8d5c70586b6140f0c43999e4ec1016f/Adafruit_ADS1015.h#L32

i need to read 2 (up to 4) ads1115 at the same controller at "same time"

soligen2010, 2019-01-13:
It can do multiple, you define one object for each device and pass the I2C address in on the constructor.  What this library doesn't do is run continuous mode.  For that, check out the soligen2010 fork, which supports more features of the chip.