adafruit/Adafruit_ADS1X15 #43: Incorrect pin assignments A0-A3 in .h file

Bwanna, 2020-01-02:
- Arduino board:  ESP32-DevKit3

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.10

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
   >Use example: 'singleended.ino"
   >Use: " Adafruit_ADS1115 ads" instead of " Adafruit_ADS1015 ads"

Observation: 
-On breadboard and testing A0-A3 between Vdd and Gnd the following occurs:
-Analog input >> Serial Printed ouput
       A0            --        A1
       A1            --        A2
       A2            --        A3
       A3            --        A0

Recommendation:
-Verify assignments.  (Sorry, I'm still learning about writing .h and .cpp files so not confident in suggesting changes to these.)

Bwanna, 2020-01-02:
The following changes in the "Adafruit_ADS1015.h" seem to work:
ORIGINAL:
    #define ADS1015_REG_CONFIG_MUX_SINGLE_0 (0x4000)  // Single-ended AIN0
    #define ADS1015_REG_CONFIG_MUX_SINGLE_1 (0x5000)  // Single-ended AIN1
    #define ADS1015_REG_CONFIG_MUX_SINGLE_2 (0x6000)  // Single-ended AIN2
    #define ADS1015_REG_CONFIG_MUX_SINGLE_3 (0x7000)  // Single-ended AIN3

CHANGED:
    #define ADS1015_REG_CONFIG_MUX_SINGLE_0 (0x5000)  // Single-ended AIN0
    #define ADS1015_REG_CONFIG_MUX_SINGLE_1 (0x6000)  // Single-ended AIN1
    #define ADS1015_REG_CONFIG_MUX_SINGLE_2 (0x7000)  // Single-ended AIN2
    #define ADS1015_REG_CONFIG_MUX_SINGLE_3 (0x4000)  // Single-ended AIN3

caternuson, 2020-01-14:
I can't recreated this issue. I used a [Feather ESP32](https://www.adafruit.com/product/3405) with an [ADS1115 breakout](https://www.adafruit.com/product/1085) running the singleended.ino example. Jumpering a wire from VDD or GND to each of the Ax inputs shows up as expected in the serial monitor output.

What ADS1115 breakout are you using?

caternuson, 2020-02-25:
Closing due to lack of response.

closed 2020-02-25 by caternuson