adafruit/Adafruit_ADS1X15 #51: Bug in lib or defect in chip ? 16 bit ADS1115 behaves like 12 bit chip

ullix, 2020-07-03:
I am using an ADS1115 with a ESP32 and see strange responses, which make me think that only 12 bits are used instead of 16. Since this lib was extended from the 12 bit ADS1015, I am wondering whether something has been forgotten in the lib for the 16 bit variant? Or is the chip defect? I have tried three different ones, all from one shipment, all with the same problem.

I am feeding some ~1.6V - half the 3.3V supply, obtained via a 10:10k divider - to channel A0 of the ADS1115 and measure single-ended with gain `GAIN_TWO`.

Most of the time I get ADC readouts of 26368 (= 1.648V). I would expect to get some jitter of at least 1 bit up and down, but the **only** other readout I get is 26352 on 2 of the 3 chips, which is lower by 16, or I get 26384 on 1 of the 3 chips, which is higher by 16.

        // #include <Adafruit_ADS1015.h>
        // Adafruit_ADS1115 ads;  /* Use this for the 16-bit version */

        // 2 out of 3 chips:
        AIN(0): 26352, Voltage: 1647.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26352, Voltage: 1647.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26352, Voltage: 1647.000 mV

        // 1 out of 3 chips
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26384, Voltage: 1649.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26384, Voltage: 1649.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26384, Voltage: 1649.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26384, Voltage: 1649.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV
        AIN(0): 26368, Voltage: 1648.000 mV

Since 26368 = 0x670**0**, 26384 = 0x671**0**, and 26352=0x66F**0** it looks like the lowest 4 bits might have been ANDed out?  I checked the lib, but I could not find the problem.

The chip itself has an imprint of 

> 92 [TI logo] BOGI

which I believe confirms it as a ADS1115.

ladyada, 2020-07-03:
@ullix what is the exact breakout you purchased?

ullix, 2020-07-03:
This one:
[https://www.amazon.de/gp/product/B07QHWLTTS/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1](url)

On the back of my chips there is an imprint "QIFEI" with a logo unknown to me. The closest to that was this on amazon: [https://www.amazon.de/sp?_encoding=UTF8&asin=B07BFVFJH6&isAmazonFulfilled=1&isCBA=&marketplaceID=A1PA6795UKMFR9&orderID=&seller=A19GB54XH1C2M3&tab=home&vasStoreID=](url); not sure if related.

ladyada, 2020-07-03:
since you're the only person with this issue, please contact the vendor first

closed 2020-07-03 by ladyada

freestyledork, 2020-11-28:
I also experience the same issue.. Did you ever get it figured out?

ullix, 2020-11-28:
Yes. Without any doubt my chips were forgeries! 12 bit ADS1015 chips very professionally disguised as 16 bit ADS1115.

The steps to detect are described here: https://forum.arduino.cc/index.php?topic=694956 
The pages are in German, maybe Google can translate. Otherwise, there is code and data sprinkled in the pages which may sufffice to let you understand.

freestyledork, 2020-11-28:
> Yes. Without any doubt my chips were forgeries! 12 bit ADS1015 chips very professionally disguised as 16 bit ADS1115.
>
> The steps to detect are described here: [...]

Thanks for the quick follow-up. I was afraid they were forges.  I tried another library and had the same 12bit results, so it's not the library. I also saw a bunch of comments from the amazon knock offs that the boards are actually 12bit ADS1015. I will be returning mine.

BBtheEE, 2020-12-28:
Yep, I just ran into the same issue.  I thought there was a bug in the library at first, but it turns out it's the chip.  I checked the markings, which look legit, but apparently they are not.

ullix, 2020-12-29:
@BBtheEE  @freestyledork Just to check whether there is a common source for the false chips: where did you purchase them? I got mine from AZDelivery via Amazon: https://www.amazon.de/gp/product/B07QHWLTTS

freestyledork, 2020-12-30:
@ullix I tried 3 different suppliers. They all ended up being the wrong chip. I purchased from HiLetgo / MakerFocus / Organizer on Amazon. I returned all 3.

ladyada, 2020-12-30:
lol think of how much money you saved! 👍

freestyledork, 2020-12-30:
> lol think of how much money you saved! 👍

All of it? Shipping is free and it doesn't cost anything to return stuff.

ladyada, 2020-12-30:
your time is worth more nothing :)