adafruit/Adafruit_ADS1X15 #56: Add ability to configure Data Rate

caternuson, 2021-02-16:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=175501

Add ability to configure Data Rate? i.e, the DRx bits in the Config Register. Looks like things are currently hard wired to a single value.

caternuson, 2021-03-25:
Added with #59

closed 2021-03-25 by caternuson

Russell108, 2022-12-10:
Calling this function data rate is not intuitive Why not call it sample rate well I am assuming this is what is being configured

Russell108, 2022-12-10:
> It reflects the nomenclature from the datasheet:
>
>  <https://user-images.githubusercontent.com/8755041/206864901-87cf1d1b-a450-43a8-b877-88f8eecbc23c.png>

I Believe I have realised this library has a  limitation

evet though the data rate is changed, it does little to speed up the delivery of data to the Arduino

Since it uses a time rather than the alarm to check if data is available to read

In my opinion, this makes the library very ineffective in real terms. 

This library does a better job. https://github.com/jrowberg/i2cdevlib