adafruit/Adafruit_ADS1X15 #63: Version 2.0 crashes at `begin()`

David-OConnor, 2021-03-31:
Hi. I've been using this library for a while with no issues, but version 2.x crashes at `begin()`. The code compiles after minor adjustments. If I revert to 1.x, (And make the appropriate module name and address location changes), it works again. How would you troubleshoot this?

Related: It's worth updating [https://learn.adafruit.com/adafruit-4-channel-adc-breakouts/arduino-code](this Adafruit guide) with the new module name `ADX1x15`, and the address going in `begin()` instead of the constructor. And adding how to set a custom address to the examples folder here.

Thank you.

caternuson, 2021-03-31:
What main board are you using?

David-OConnor, 2021-03-31:
Arduino Uno

caternuson, 2021-03-31:
Just tested here and it works OK. The 2.x updates did have some breaking changes, so you may need to make some minor changes to any existing code. To sanity check, try using one of the examples from this library.

This was using the single ended example:
https://github.com/adafruit/Adafruit_ADS1X15/blob/master/examples/singleended/singleended.ino
with A0 and A2 tied to the voltage divider. A1 tied to GND. A3 tied to 5V.

[image]

[image]

David-OConnor, 2021-04-01:
Thank you very much for the detailed info. Th example code works for me. I'm using this library as a dependency in the Arduino driver for a pH circuit, and am still unable to get v2.x working, but can't figure out why. My best guess is that this is related to the busio dependency change. I'm not getting an error - just a hang, and/or gibberish printed to the console. I'll troubleshoot more, and keep you posted.

ladyada, 2021-04-01:
you're probably running out of RAM. you'll need to update to a board with more RAM memory.

caternuson, 2021-04-06:
Closing due to lack of response. Since the library example sketch works, this may either be the RAM issue or needing some slight code updates to deal with minor (breaking) changes in 2.x release.

closed 2021-04-06 by caternuson