adafruit/Adafruit_ADS1X15 #65: 2.1.1 in 2/3 gain mode, values loose resolution

OZ2CPU-thomas, 2021-04-15:
at voltages over 5.7V to 6.1V the values loose the last digit of analog resolution
tha tlast value become 0
it is supposed to handle 0-6,1V in a smooth full resolution way,
at input voltages under 5.7V the data do contail values in the last digit
my IC is ADS1115 so it is 16 bit

caternuson, 2021-04-15:
The 6.144V FS range is for scaling only. The actual input range should not exceed VDD+0.3V. VDD absolute max is 5.5V.

[image]

[image]

OZ2CPU-thomas, 2021-04-15:
THANKS !! this could explain it.. still a quite odd error, since it handle +/- voltages too ??

closed 2021-04-15 by ladyada