adafruit/Adafruit_ADS1X15 #66: Adafruit_ADS1X15 Library

cesartayoun, 2021-06-05:
i am using an ADS1115 16 bits with ESP32, i noticed that the ADS library does not have a function to check the availability of the sensor, hence if the sensor is not available the ESP32 will freeze.
any workaround for such problem ? thanks

ladyada, 2021-06-07:
@caternuson plz make
https://github.com/adafruit/Adafruit_ADS1X15/blob/master/Adafruit_ADS1X15.cpp#L65
return the value from the i2cdev begin so it can be verified that the addr exists?

caternuson, 2021-06-07:
@ladyada yah, should just do that regardless. i'll pr it real quick.

@cesartayoun can you share code that demonstrates the issue? wondering if the actual call to `i2cdev.begin()` is freezing?

caternuson, 2021-06-07:
Possibly related to #64

caternuson, 2021-06-07:
@cesartayoun Please update to the 2.2.0 release of the library:
https://github.com/adafruit/Adafruit_ADS1X15/releases/tag/2.2.0
and see if that helps.

cesartayoun, 2021-06-08:
Hello, i tried the singleended example, no particular code to share; i will update the library and try.. thanks

cesartayoun, 2021-06-09:
Hi again, problem solved, i modified the library as instructed above. thank you

closed 2021-06-09 by ladyada