adafruit/Adafruit_ADS1X15 #69: fatal error: Adafruit_ADS10s15.h: No such file or directory

allrobot, 2021-07-08:
error:
[image]
This is an online VS Code installation and I'm trying to boot into the main file.
Ads1015 data:
[image]
How to solve it?

allrobot, 2021-07-08:
I tried to reference other library files and uploaded them normally, only ADS1015 had problems

ladyada, 2021-07-08:
install https://github.com/adafruit/Adafruit_BusIO

closed 2021-07-08 by ladyada

allrobot, 2021-07-09:
> install https://github.com/adafruit/Adafruit_BusIO

Thanks, the problem was successfully resolved!
[image]