adafruit/Adafruit_ADS1X15 #72: Offset in NodeMCU when analog inputs = 0V

DaniMzT, 2021-11-30:
Hello,
First of all, thanks for the library :) I'm using it with 3 analog inputs connected to an ADS1115, which sends the data to a NodeMCU. I'm using version 2.2.0. 
My problem is that there is a voltage offset when I read the inputs in NodeMCU, even using the example "singleended" from the library. The voltage of the inputs is around 90 mV (measured in relation to ADS1115's GND) but in the serial monitor I'm reading 0.4V.
In my particular code, I've also tried it with gain = ONE (+-4V reference, so 0.125 mV resolution) but the results are the same. 
Here a video I've just recorded as an evidence: https://www.youtube.com/watch?v=TMURcEnpoeA 
As you can see, my multimeter displays .08 (80 mV) for A0, A1 and A2. Don't pay much attention to A3, because it was connected to an open point. 
Do you know if there is a known issue and there is a solution? Or should I go on by adding an offset in my code?
Thanks in advance!
Kind regards

DaniMzT, 2021-12-06:
I've repeated the tests with another ADC, but the results are as wrong as with the other ADC. These are the ADCs used:
https://www.amazon.es/gp/product/B01N8ODF5C/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1
https://es.aliexpress.com/item/32882005440.html

soligen2010, 2021-12-09:
I think you found your issue.  OP Amps typically have an offset.  Some op amps have a way to trim out the offset.  Also, if it is not powered with both positive and negative supply, it is very common that the op amp will not go all the way to zero (even if it is "rail to rail") - this fits your observations quite well and I think is what is happening to you.  I don't think your issue is with the ADC or the library, but is in the electronics before it.   I suggest getting rid of the op-amp unless you want to do the detailed analysis and extra engineering to compensate for the offset and ensure the output can go to zero.

caternuson, 2021-12-10:
To verify this is a library issue, you'll need to remove the ADS1115 from the custom setup.

[image]

A good test setup would the Feather driving the ADS115 with a trim pot voltage divider going into the ADS, all staged on a breadboard.

DaniMzT, 2021-12-16:
> I think you found your issue. OP Amps typically have an offset. Some op amps have a way to trim out the offset. Also, if it is not powered with both positive and negative supply, it is very common [...]

Exactly that, thank you! The problem was my op.amo. wasn't rail to rail. I've set specific voltage on ADS1115 inputs directly and it was working exactly. I'm not expert on electronics so I didn't know this concept of rail-to-rail. The low output voltage of my amp (LP324N) is 0.7V when Vcc is 5V (datasheet). In my case, it was 0.4 V with Vcc=3.3.
 Sorry for bothering, the library works very well. I'll search a more accurate amp!

closed 2021-12-16 by DaniMzT

caternuson, 2021-12-16:
@DaniMzT Thanks for reporting back and closing the issue.

@soligen2010 Good job finding the issue!

DaniMzT, 2021-12-16:
> @DaniMzT Thanks for reporting back and closing the issue.
> 
> @soligen2010 Good job finding the issue!

Thank you guys for helping and developing libraries like this 👍 What a good job!