adafruit/Adafruit_ADS1X15 #84: ADS1X15.h .cpp doesn't work with Seeduino Xiao

jritz1349, 2023-07-30:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

jritz1349, 2023-07-30:
When I compile using ADS1X15.h with Seeduino XIAO I get errors:

error: no matching function for call to 'Adafruit_ADS1X15::readADC_SingleEnded()' adc1 = ads1115.readADC_SingleEnded();

but clearly in the ADS1X15.cpp there is a readADC_SingleEnded function.

caternuson, 2024-02-03:
Which specific XIAO board is being used? There are many.

What Board Support Package is being used?

What board is being selected in the Ardunio IDE?

jritz1349, 2024-02-05:
Hello,

Thank you for answering my questions. This was the Seeduino development
board.
I was able to resolve this now. I believe I just used the command
"readADC_SingleEnded(0)" as the port 0 was used.

-John Ritz

caternuson, 2024-02-06:
The Xiao is entire series, not a single product:
https://www.seeedstudio.com/xiao-series-page

Closing since this sounds resolved. If there is still an issue, please provide specifics on the hardware being used.

closed 2024-02-06 by caternuson

jritz1349, 2024-02-09:
Yes, its the Seeed Studio XIAO SAMD21

Thanks.