adafruit/Adafruit_ADS1X15 #92: Library does not properly put ADS1115 to sleep in single-ended mode, uses full 300uA+ 24/7

moeburn, 2024-06-18:
I am testing this with a multimeter.  Comparing this sketch:

https://github.com/adafruit/Adafruit_ADS1X15/blob/master/examples/singleended/singleended.ino

...with this nearly identical sketch from a different ADS1115 library:

https://github.com/wollewald/ADS1115_WE/blob/master/examples/Single_Shot/Single_Shot.ino

The Adafruit library results in the multimeter showing ~320uA constantly.  The other _WE library shows the multimeter reading 0.2uA most of the time, and then briefly jumping up a bit during each read before going back down to near-zero.  This is how the ADS1115 is supposed to behave in single-ended mode.

I cannot confirm if this behaviour is the same on genuine Adafruit brand modules.  I can only say that this library is no good if you are using 3rd-party ADS1115 modules, if you are expecting full power savings in single-ended mode.

caternuson, 2024-06-18:
What ADS1115 hardware is being used for testing?

Is the only difference between the setups the Arduino library being used?

moeburn, 2024-06-28:
I've tried this on all the various ADS1115 modules I have.  Correct, the only difference is changing libraries, and suddenly the current properly drops to 0 after a single ended reading.