adafruit/Adafruit_ADS1X15 #96: Unable to verify correct I2C communication

Giermann, 2025-03-03:
The whole class is very vulnerable for communication problems on I2C bus!
The methods writeRegister() and readRegister() are using m_i2c_dev->write() and m_i2c_dev->read() which return a bool value, but these are never evaluated.

In this case, one would never recognize if read voltages are actual readings or simply the last successful reading or even the last (un?)successful writing.

It would be easy to convert writeRegister() to return the bool result of m_i2c_dev->write() and to handle this or simply also return the bool result in startComparator_SingleEnded() and startADCReading().

For readRegister() it would require some more refactoring, also read #75 for some thoughts on this.

caternuson, 2025-03-03:
Are you experiencing an actual issues? If so, please provide more details about the hardware setup, code being run, and what is happening.

Or is this just observation of source code?

Giermann, 2025-03-03:
I do have an actual issue with some other sensor and need to implement some error handling in my appliance in case of communication errors. During that I wanted to add communication check for the ADS1115 as well, which is currently impossible with this library.

My hardware setup:
I run a current sensor attached to an ADS1115. Reading that sensor, I decide whether do adjust some settings or finally cut-off a relay when excessing a specified maximum current. If (for any reason) my I2C connection does not work anymore, I assume the last read current to by still valid, because I do not get any error while reading the (old) buffer...

But I'll create a PR in the next days, a fix should not be too complicated.